ros-perception/vision_msgs #15: Some questions about vision_msgs project

chaoli2, 2018-08-07:
@Kukanani 
Based on your kindly suggestion [here](https://github.com/intel/ros2_object_msgs/issues/1), we are now evaluating the efforts and benefits of moving current [object_msgs ](https://github.com/intel/ros2_object_msgs) to vision_msgs. This migration may influence some existing projects(like [ros2_intel_movidius_ncs ](https://github.com/intel/ros2_intel_movidius_ncs), [ros2_object_analytics](https://github.com/intel/ros2_object_analytics), etc.) and most future projects.  We still need your help for some questions before we could make final decision:

1. We found that [vision_msgs](http://wiki.ros.org/vision_msgs) is already available for ROS. Is there any schedule to merge this message type to ROS2 core, since most of current projects we are working on are for ROS2?

2. Is the debian package of this project available in the future? Some of our projects are being integrated into ROS2 core and the message project is one of the dependencies. Since OSRF only accepts debian package dependencies in ROS2 core project, we'd like to know your plan about this. 

Vision_msgs is really a valuable project to help unify the interface of computer vision and object detection tasks. Thanks very much for working it out and hope we have more chances to co-work on CV and ROS/ROS2 :)

adam-diligent, 2018-08-09:
Hello @chaoli2, great to hear from you!
1.  There is already a ROS2 version of the package, which is on the `ros2` branch of the repo. I don't think I've released it yet, though! This is definitely something I'll look into
2. The debians for the ROS1 version of the package are already available (i.e. `sudo apt install ros-kinetic-vision-msgs` should work). It sounds like you are specifically asking about ROS2 debians, right? Those aren't available yet.

I haven't released a package into ROS2 yet, but I think the procedure is the same as in ROS1, so if I did a ROS2 release it should make it into the debians in the next package sync. I'll look into this as well--I'm excited that this package is seeing some wider adoption and I'm definitely willing to support those efforts.

I'll try to get you an update on this as soon as possible. Looking forward to see cool integrations in the future!

Kukanani, 2018-08-09:
Hello @chaoli2, great to hear from you!

    There is already a ROS2 version of the package, which is on the ros2 branch of the repo. I don't think I've released it yet, though! This is definitely something I'll look into
    The debians for the ROS1 version of the package are already available (i.e. sudo apt install ros-kinetic-vision-msgs should work). It sounds like you are specifically asking about ROS2 debians, right? Those aren't available yet.

I haven't released a package into ROS2 yet, but I think the procedure is the same as in ROS1, so if I did a ROS2 release it should make it into the debians in the next package sync. I'll look into this as well--I'm excited that this package is seeing some wider adoption and I'm definitely willing to support those efforts.

I'll try to get you an update on this as soon as possible. Looking forward to see cool integrations in the future!

Kukanani, 2018-08-10:
I did more digging and realized that I already released for ROS2 Ardent some time ago: [link](https://github.com/ros2/rosdistro/commit/32d00df3303725c31e3d1dd706bf250638b910f7). Does this suit your team's needs?

chaoli2, 2018-08-10:
@Kukanani Thanks for your quick response. I saw its commit history to ros2 build farm. I'd like to check corresponding debian package in few days. Will you let know if any question 👍

chaoli2, 2018-11-20:
We temporarily add our [object_msgs](https://github.com/ros2/rosdistro/pull/334) into Crystal release so as to satisfy dependency of other projects. For long term consideration, we'd like vision_msgs is supported in every future ROS2 realease :)

Kukanani, 2018-11-20:
Thank you for releasing! As someone who does not use ROS2 on a regular basis it is sometimes difficult for me to remember to release to a new version. I do plan to continue support. Thank you for the reminder!