ros-perception/vision_msgs #26: Add ObjectHypothesisArray

mistermult, 2019-08-05:
What's the reason for all this messages? We want to detect objects in images or point clouds to build a model of the real world. The real world exists even if there are there is no image taken of it. So we need a message that only contains objects but does not reference images or point clouds.
Applications:
* We put objects on the table and measure the coordinates. This is the ground truth data without images. Later we do object detection and compare the result with the ground truth.
* We have ground truth data of fixed objects and want to visualize them alongside of detected objects.

Therefore I suggest ObjectHypothesisArray.msg
```
Header header
ObjectHypothesisWithPose[] objects
```

mintar, 2019-08-05:
Why not simply use a Detection3DArray for ground truth? In my experience, it's really useful to use the same format for the ground truth and the detections:

* you can re-use all existing visualizations etc.
* It's easy to compare actual values (= detections) and expected values (= ground truth)

Kukanani, 2019-08-05:
This package is called `vision_msgs` because I designed it for use with systems that detect and classify using visual information. So I never planned to represent the world in a way that was not informed by visual information.

mistermult, 2019-08-05:
The problem is that one has to define an arbitrary bounding box and one can set multiple hypothesis per bounding box. But I understand that this package only deals with vision messages. So I understand that this is not added.

closed 2019-08-06 by Kukanani