ros-perception/vision_msgs #27: Clarify size in BoundingBox2D, BoundingBox3D.

mistermult, 2019-08-05:
The code says that it is:
```
# The size of the bounding box, in meters, surrounding the object's center
#   pose.
geometry_msgs/Vector3 size

# The size (in pixels) of the bounding box surrounding the object relative
#   to the pose of its center.
float64 size_x
float64 size_y
```

Is this width the distance to the center (halfsize) or the extent in x-direction?

mintar, 2019-08-05:
It's the size, i.e., the extent in x-/y-direction. I agree that the comment could be clearer - what does the bounding box center have to do with its size?

mistermult, 2019-08-05:
I guess these are oriented bounding boxes and the orientation is saved in the center. That means if you rotate by pi/2 around z, the extent in x/y direction is interchanged :)

SteveMacenski, 2020-12-05:
Now(-ish) would be a good junction to clarify this. Would something like `The absolute size (in pixels)...` fix this issue? I think that makes it clear its the total size, not the half size.

Kukanani, 2020-12-20:
`absolute size` or `total size` would seem to clear it up for me. We should also probably clarify that `relative to the pose of its center` includes orientation (i.e. the x and y axes are expressed in the frame defined by the the `center` field).

SteveMacenski, 2020-12-20:
I don't think we need to change the name of the field in the message, I think we just need to add a comment to clarify what the value means.

PR #51 (2021-01-05): Clarify comment for size fields in bounding box messages
Closes #27. See that issue for details.
closed 2021-01-05 by Kukanani