ros-perception/vision_msgs #37: Query regarding usage of vision_msgs::msg::Detection2DArray

VeCAD, 2020-07-21:
I can see the darknet detector values (bbox, object id) is being published using ros2 topic echo /topic. But I'm having issue pulling the detector values. 

At the subscriber:
`const vision_msgs::msg::Detection2DArray::SharedPtr det_msg`

I can pull the header info as below. 
` std::cout << det_msg->header.frame_id << std::endl;`

Is there any example that uses Detection2DArray at subscription side?

Kukanani, 2020-07-29:
Hi, I am not sure what code you are using to generate the detections, but if this package is being used as intended, then I think you're looking for  `det_msg->detections[0].results.id` and `det_msg->detections[0].results.score`.

The `id` field should allow you to look up the results in a lookup table that should be supplied either in the dataset or by the detector. See https://github.com/Kukanani/vision_msgs/blob/kinetic-devel/msg/ObjectHypothesisWithPose.msg#L5 for details.

Kukanani, 2020-09-24:
Closing due to inactivity.

closed 2020-09-24 by Kukanani