ros-perception/vision_msgs #39: [ros2] ros1_bridge can't map ObjectHypothesis between ROS 1 and ROS 2

norro, 2020-09-23:
ros1_bridge can't map ObjectHypothesis between ROS 1 and ROS 2 because of two conflicting types in ObjectHypothesis' `id` field. The field is `int64` in ROS 1 and `string` in ROS 2.
During compilation of ros1_bridge, the automatic mapping attempt will fail with the following error:

```
.../ros1_bridge/generated/vision_msgs__msg__ObjectHypothesis__factories.cpp: In static member function ‘static void ros1_bridge::Factory<ROS1_T, ROS2_T>::convert_2_to_1(const ROS2_T&, ROS1_T&) [with ROS1_T = vision_msgs::ObjectHypothesis_<std::allocator<void> >; ROS2_T = vision_msgs::msg::ObjectHypothesis_<std::allocator<void> >]’:
.../ros1_bridge/generated/vision_msgs__msg__ObjectHypothesis__factories.cpp:79:26: error: cannot convert ‘const _id_type {aka const std::__cxx11::basic_string<char>}’ to ‘vision_msgs::ObjectHypothesis_<std::allocator<void> >::_id_type {aka long int}’ in assignment
   ros1_msg.id = ros2_msg.id;
                          ^~
make[2]: *** [CMakeFiles/ros1_bridge.dir/generated/vision_msgs__msg__ObjectHypothesis__factories.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
.../ros1_bridge/generated/vision_msgs__msg__ObjectHypothesisWithPose__factories.cpp: In static member function ‘static void ros1_bridge::Factory<ROS1_T, ROS2_T>::convert_2_to_1(const ROS2_T&, ROS1_T&) [with ROS1_T = vision_msgs::ObjectHypothesisWithPose_<std::allocator<void> >; ROS2_T = vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> >]’:
.../ros1_bridge/generated/vision_msgs__msg__ObjectHypothesisWithPose__factories.cpp:87:26: error: cannot convert ‘const _id_type {aka const std::__cxx11::basic_string<char>}’ to ‘vision_msgs::ObjectHypothesisWithPose_<std::allocator<void> >::_id_type {aka long int}’ in assignment
   ros1_msg.id = ros2_msg.id;
                          ^~
```

This probably requires a custom mapping rule (which is possible for ros1_bridge).

norro, 2020-09-23:
Probably relates to this recent commit: https://github.com/Kukanani/vision_msgs/commit/82dda4ab3affb6e7c142054f86c7a897f0b8b41e

mintar, 2020-09-23:
To clarify: The field is `int64` in ROS1 kinetic and melodic, and `string` in ROS1 noetic and ROS2.

norro, 2020-09-23:
I'd say that in a setup where it bridges the old (`int64`) world and the new (`string`) world, it is fair to assume that IDs are (string-ified) numbers.

mintar, 2020-09-23:
> I'd say that in a setup where it bridges the old (int64) world and the new (string) world, it is fair to assume that IDs are (string-ified) numbers.

Fair enough. The bridge could just try whether the conversion works and throw an exception if the string cannot be converted to an int.

norro, 2020-09-23:
Might be a delicate situation, because custom mapping rules for the ros1_bridge are supposed to live in the ROS 2 package, IMHO, but if it is required or not is depending on the ROS 1 version in use.

Kukanani, 2020-09-24:
Thanks for the bug report!

Unfortunately I don't have the bandwidth to work on this and come up with a good solution right now. I'm happy to review PRs or advocate for a PR on the bridge repo, though. I'm going to create a "help wanted" label and apply it to this issue.

Also, note that while Melodic is not EOL until 2023, the ROS2 distros that also target Ubuntu 18 (Dashing and Eloquent) are EOL in May 2021 and November 2020. So any fix for these distros that involves ros2_bridge fixes would have to come in before then.

SteveMacenski, 2026-04-02:
ROS 1 is EOL and will not receive further support