ros-perception/vision_msgs #40: Published message is not received

vvigilante, 2020-10-09:
Hello!
Tested with ROS melodic on Ubuntu 18.04

**rostopic** info shows that the topic is published and subscibed as expected.
**rostopic hz** shows that the topic is being published with a frequency of ~10hz, as expected.
**rostopic echo is completely silent**, so is the subscriber node that I wrote.

It works if I change the type of the published topic to String or to Image, it does not work with Detection2D or  Detection2DArray

My code looks like this:
```python
    from time import sleep
    import rospy
    rospy.init_node('t1')
    pub = rospy.Publisher('detection', Detection2DArray, queue_size=10)
    while True:
        message = Detection2DArray()
        pub.publish(message)
        sleep(0.1)
```

Kukanani, 2020-10-09:
Please post the code of the non-working subscriber and the exact commands
you are running in your terminal. Thanks

vvigilante, 2020-10-09:
```bash
rostopic list
rostopic hz /detection
rostopic bw /detection
rostopic echo /detection
```
```bash
rostopic info detection
``` 
gives
```
Type: vision_msgs/Detection2DArray

Publishers: 
 * /t1 (http://ubuntu:46019/)

Subscribers: 
 * /t2 (http://ubuntu:35623/)
```

The code for the subscriber is the following.

```python
import rospy
from vision_msgs.msg import Detection2DArray

rospy.init_node('t2')

def rcv_detection(msg):
    rospy.loginfo('detection here')

sd = rospy.Subscriber("detection", Detection2DArray, rcv_detection)
rospy.spin()

```

It stays completely silent.

Kukanani, 2020-10-10:
The code you posted is incomplete. Your publisher does not `import Detection2DArray`.

In general, if `rostopic hz` works and `rostopic echo` does not, this is probably an issue with your machine setup or network transport, not the message type itself. I would investigate other issues such as
* your ROS environment variables (`ROS_IP` or `ROS_HOSTNAME` are common problems)
* checking that both terminals have `source`d the workspace with `vision_msgs` in it
* checking that you can run an interactive `python` interpreter and `import vision_msgs`
* machine setup (are you running these nodes on different machines or do you have a remote master?)
* `roswtf`
* computer networking configuration

I was **unable** to reproduce the bug on ROS melodic, Ubuntu 18.04, latest `vision_msgs` on `melodic-devel` branch, using the following code. I get continuous `detection here` log messages on the subscriber. I verified that the following example works on both my own desktop machine and in a `ros:melodic` docker container. Therefore, I am going to close the issue as `wontfix` since it does not appear to be a problem with this package. If you find more detailed information that specifically points to 

Note that because you do not check for the `rospy.is_shutdown()` state, it is "difficult" to kill `publisher` (Ctrl-/ not Ctrl-C in `bash`)

publisher
```
#!/usr/bin/python

from time import sleep
import rospy
from vision_msgs.msg import Detection2DArray

rospy.init_node('t1')
pub = rospy.Publisher('detection', Detection2DArray, queue_size=10)
while True:
    message = Detection2DArray()
    pub.publish(message)
    sleep(0.1)
```

subscriber
```
#!/usr/bin/python

import rospy
from vision_msgs.msg import Detection2DArray

rospy.init_node('t2')

def rcv_detection(msg):
    rospy.loginfo('detection here')

sd = rospy.Subscriber("detection", Detection2DArray, rcv_detection)
rospy.spin()
```

closed 2020-10-10 by Kukanani

vvigilante, 2020-10-12:
Dear Kukanani,
thank you for your help.

roswtf gives
```
ERROR The following nodes should be connected but aren't:
 * /t1->/t2 (/detection)
```

Setting `ROS_ID=127.0.0.1` does not help.

The problem only happens if I `apt install ros-melodic-vision-msgs`.

If I clone the repository, `checkout melodic-devel` and `catkin build`, everything works as expected