ros-perception/vision_msgs #41: Interest in moving this to be official vision_msgs in ros-perception org?

SteveMacenski, 2020-12-03:
Hi,

I see this package being used as a standard interface for detection / AI models in ROS. It would be great if this was more "official" then as a `ros-perception/vision_msgs` package so that it can be "official" vision interfaces for ROS. I have the permissions required to help make this happen if you like!

Steve

mintar, 2020-12-04:
I fully agree, that would be a great idea! I probably don't have to tell you this, but make sure to actually *move* the repo, not fork/copy it. That way, all the links (and even git clone) will keep working with the old URL (GitHub does a HTTP redirect).

Kukanani, 2020-12-04:
Hi Steve,

I fully support moving this package to `ros-perception`. The only reason it didn't get created there originally was that we weren't sure how much community adoption it would see.

SteveMacenski, 2020-12-04:
Awesome! I think the easiest way to handle this is the following
- Transfer the repo to me, since you don't have admin on `ros-perception`
- I'll transfer it to `ros-perception`. I'm not sure what all happens with settings in that case, so I'll make sure you still have admin over there in case it removes it for some reason

Kukanani, 2020-12-05:
Sounds good. Transfer initiated.

SteveMacenski, 2020-12-05:
I see the request, github says it might take a bit to get over to my side so I can move it to `ros-perception`. Hopefully should happen tonight, if not over the weekend. I'll follow back here when its done.

SteveMacenski, 2020-12-05:
[image]

Well this is truly unique. Its on my account by no `settings` icon for me to move it around... mhm maybe takes a bit for all the permissions to update?

Kukanani, 2020-12-05:
What have you done to my repository? 😛 
[image]

SteveMacenski, 2020-12-05:
It'll all work itself out, don't worry :smile: I have no interest in keeping it. If anything else, I'll email someone at GitHub on Monday. They're usually pretty good about working through issues like this quickly. This isn't the first time I've done the "repo-shuffle" to move things into `ros-*`, but first time it didn't automatically give me a `settings` panel...

Kukanani, 2020-12-05:
Sounds good! Hopefully there are just some database updates that need to propagate around. Please keep me posted.

SteveMacenski, 2020-12-07:
Nothing changed now on Monday, I contacted GitHub for support.

SteveMacenski, 2020-12-08:
https://github.com/ros-perception/vision_msgs

Now on `ros-perception`!

closed 2020-12-08 by SteveMacenski