ros-perception/vision_msgs #42: Messages to contain uncertainty information

SteveMacenski, 2020-12-03:
In reviewing the messages, I see they are missing measures of uncertainty in the detections / output values. It would be valuable information to communicate to downstream uses of these messages.

mintar, 2020-12-04:
Actually, that field already exists. It's called `score`:

https://github.com/Kukanani/vision_msgs/blob/82dda4ab3affb6e7c142054f86c7a897f0b8b41e/msg/ObjectHypothesis.msg#L8-L10

Kukanani, 2020-12-04:
Martin's right. We also provide covariance information for the detected pose: 

https://github.com/Kukanani/vision_msgs/blob/82dda4ab3affb6e7c142054f86c7a897f0b8b41e/msg/ObjectHypothesisWithPose.msg#L20

SteveMacenski, 2020-12-04:
I was primarily looking at the bounding box messages, which do not contain any score / covariance information. After more looking, it seems that those messages are supposed to be a component of the Detection / Classification messages so that's all good.

Thanks

closed 2020-12-04 by SteveMacenski