ros-perception/vision_msgs #46: ObjectHypothesisWithPose array

tp4348, 2020-12-15:
Why is there no array for ObjectHypotesisWithPose? I understand anybody can do it themselve, but that doesn't make this type of message a standard one.

mintar, 2020-12-15:
`ObjectHypothesisWithPose` is intended to always be used inside a `Detection2D` or `Detection3D` message.

You could say that `Detection2D`/`Detection3D` *are* arrays of `ObjectHypotesisWithPose`, plus some other useful information (header, bbox, source data and tracking id).

SteveMacenski, 2020-12-15:
But this ticket does bring up a good point that maybe that's context we can put into the msg as a comment

Kukanani, 2020-12-20:
Closed by #46
@tp4348 feel free to re-open if this doesn't address your needs.

PR #49 (2020-12-20): Clarify: ObjectHypothesis[] ~= Classification
https://github.com/ros-perception/vision_msgs/issues/46 requested Array message types for ObjectHypothesis and/or ObjectHypothesisWithPose. As pointed out in the issue, these already exist in the form of the `ClassificationXD` and `DetectionXD` message types.

Closes #46
closed 2020-12-20 by Kukanani