ros-perception/vision_msgs #54: Incompatibility between messages for different ROS distros

silasalves, 2021-01-15:
I noticed that some messages (in my case, Detection2D) is different on ROS Melodic and Noetic:

Melodic | Noetic
------------ | -------------
std_msgs/Header header<br>vision_msgs/ObjectHypothesisWithPose[] results<br>vision_msgs/BoundingBox2D bbox<br>sensor_msgs/Image source_img<br><br><br> | std_msgs/Header header<br>vision_msgs/ObjectHypothesisWithPose[] results<br>vision_msgs/BoundingBox2D bbox<br>sensor_msgs/Image source_img<br>**bool is_tracking**<br>**string tracking_id**

Is there a reason for omitting `is_tracking` and `tracking_id` on Melodic? I installed the package using APT, but I could also clone this repo in my workspace in case I need the tracking messages in the future.

Kukanani, 2021-01-15:
`is_tracking` and `tracking_id` were added after `melodic` was released, and we have a general rule in the ROS community to not change message definitions after a distro release, because this causes md5 compatibility issues (the message has to be rebuilt everywhere). Therefore, the fields were added in `noetic` but not earlier versions.

Please note that the tracking fields are going away or being renamed in newer versions of this package (https://github.com/ros-perception/vision_msgs/pull/50)

silasalves, 2021-01-15:
Thanks a lot for the heads-up!

closed 2021-01-15 by silasalves