ros-perception/vision_msgs #55: Rolling Release

lukicdarkoo, 2021-04-09:
Are there any plans on releasing this package to Rolling?

Kukanani, 2021-04-10:
Sure thing. I’ll look into it on Monday. Thanks for the ping!

Kukanani, 2021-04-13:
I released to rolling, but was unable to get `bloom` to automatically create the PR in `rosdistro` after several tries.

https://github.com/Kukanani/vision_msgs-release

I keep getting various 403 errors, such as the following:
```
Aborting pull request: HTTP Error 403: Forbidden (https://api.github.com/repos/ros/rosdistro/forks): None
```

I'm not really interested in filling out the lengthy rosdistro PR template myself, and more generally I'd like to get this problem resolved on my machine. @SteveMacenski any tips? My personal access token has full repo and workflow access.

SteveMacenski, 2021-04-13:
I don’t know, haven’t seen that before

Kukanani, 2021-04-14:
Tracking the problem at https://github.com/ros-infrastructure/bloom/issues/632. If there are no updates there for a few days I'll make the PR manually.

SteveMacenski, 2021-04-14:
Opening the PR is the last step, so it should be ready to go with a body block to just copy paste in if that was the only issue

Kukanani, 2021-04-16:
https://github.com/ros/rosdistro/pull/29188

closed 2021-04-16 by SteveMacenski