ros-perception/vision_msgs #59: Galactic Release

lukicdarkoo, 2021-07-19:
Like in #55, we also need a Galactic release

BTW, thank you for the Rolling release :)

SteveMacenski, 2021-07-19:
@Kukanani I was going to do it, but realized that the release repo was still under your github user ID.

If you'd like to share that responsibility, you could transfer the repo to me and then I'd transfer it to https://github.com/ros2-gbp

Kukanani, 2021-07-21:
We transferred `vision_msgs-release` to `ros2-gbp` (see https://github.com/ros/rosdistro/pull/29311#discussion_r646969374). Are you referring to a different release repo?

Kukanani, 2021-07-21:
Created the 3.0.1 release to capture #58.

rosdistro PR for Galactic is here: https://github.com/ros/rosdistro/pull/30278

This was much easier than the "two steps forward, one step back" that was the rolling release (#55).

LIDAR manufacturers thwarted....for now! Seriously though, @SteveMacenski let me know if you have permissions on the [release repo](https://github.com/ros2-gbp/vision_msgs-release), and if not, we should reach out to Open Robotics folks to correct it. I don't have full admin control over the new release repo.

SteveMacenski, 2021-07-21:
I do, sorry for some reason when I went to run it it looked like it was still on your repo. I do have access to anything in ros2-gbp

closed 2021-07-21 by SteveMacenski