ros-perception/vision_msgs #62: About catkin_make question

ferrari700, 2021-12-11:
Hello, I am sorry to disturb you. The following error occurred when I used catkin_make in a workspace. I am sure that there is a vision_msgs folder in my src folder. How should I solve it? Thank you so much!
[image]

s-sarahSM, 2021-12-20:
@Kukanani 
I have the same problem mentionned by @ferrari700 
I am working on Nvidia jetson AGX Xavier , ROS Melodic and Ubuntu 18.04
Thank you if you can help

Kukanani, 2021-12-20:
Hello, please provide steps to produce a minimal working example, ideally using a docker container.

igillespie, 2022-01-28:
Just a reminder to clone the relative branch, I'm on ros2-galactic, so I used 
`git clone https://github.com/ros-perception/vision_msgs.git -b galactic` 
It compiled fine. I made the mistake of not specifying the branch the first time and had build issues.

closed 2022-08-24 by SteveMacenski