ros-perception/vision_msgs #68: Galactic BoundingBox3DArray message not generated for Python

vinnnyr, 2022-03-20:
In Galactic, `BoundingBox3DArray` does not show up as a generated message for python, for neither the apt binary, nor when building from source:

```
>>> from vision_msgs.msg import BoundingBox3DArray
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'BoundingBox3DArray' from 'vision_msgs.msg'
```

This occurs in a fresh docker container too, and could be replicated there: https://asciinema.org/a/vtbiRQtrlSXIvhR25SM7cMLS7

If you look at the `__init__.py` for the message (for binary located here: `/opt/ros/galactic/lib/python3.8/site-packages/vision_msgs/msg/__init__.py`) you can see that the message is not generated.  I am not sure why that may be yet.

```
from vision_msgs.msg._bounding_box2_d import BoundingBox2D  # noqa: F401
from vision_msgs.msg._bounding_box3_d import BoundingBox3D  # noqa: F401
from vision_msgs.msg._classification import Classification  # noqa: F401
from vision_msgs.msg._detection2_d import Detection2D  # noqa: F401
from vision_msgs.msg._detection2_d_array import Detection2DArray  # noqa: F401
from vision_msgs.msg._detection3_d import Detection3D  # noqa: F401
from vision_msgs.msg._detection3_d_array import Detection3DArray  # noqa: F401
from vision_msgs.msg._object_hypothesis import ObjectHypothesis  # noqa: F401
from vision_msgs.msg._object_hypothesis_with_pose import ObjectHypothesisWithPose  # noqa: F401
from vision_msgs.msg._vision_info import VisionInfo  # noqa: F401
```

vinnnyr, 2022-03-20:
Ok nevermind, I realize what happened. That message is present in the `ros2` branch, but not the `galactic` branch, and therefore I guess it won't be released for `galactic`.

vinnnyr, 2022-03-20:
well maybe I can re-open this issue just to ask -- _can_ it be cherry-picked onto the `galactic` branch / released for the next galactic sync, or does it break some sort of rule?

Kukanani, 2022-03-20:
Actually, [the file](https://github.com/ros-perception/vision_msgs/blob/galactic/msg/BoundingBox3DArray.msg) does exist in `galactic`, but it's not in the CMakeLists.txt [message files list](https://github.com/ros-perception/vision_msgs/blob/76d8ecd16e5aa2dbfcbd279077ee5f09330958cf/CMakeLists.txt). Or at least, it wasn't, until I just pushed it a few minutes ago. Now visible [here](https://github.com/ros-perception/vision_msgs/blob/galactic/CMakeLists.txt#L21).

Kukanani, 2022-03-20:
Please let me know if this works for you and we can create another release.

And just FYI, my policy (which I believe is standard for most ROS message packages) is to not make any changes to existing messages on a given distro (because this will change md5sums), but adding *new* messages is no problem.

vinnnyr, 2022-03-21:
yes this works for me now -- thank you!

Kukanani, 2022-03-21:
new release created (3.0.2), PR here: https://github.com/ros/rosdistro/pull/32488

closed 2022-03-21 by Kukanani

gbr1, 2022-08-04:
Hi!
It seems this issue is present also in foxy.

I'm not able to use BoundingBox2DArray and BoundingBox3DArray.
Thank you.