ros-perception/vision_msgs #69: BoundingBox2DArray doesn't exist in noetic release (apt)

igor-markovic, 2022-04-07:
[The wiki](http://wiki.ros.org/vision_msgs) shows `BoundingBox2DArray` as being an available message in noetic.

However, when installing `ros-noetic-vision-msgs` via apt, the message definition is missing. It looks like the [release repo](https://github.com/ros2-gbp/vision_msgs-release/branches/all?query=noetic) last updated the noetic branch in 2020, which was before the array msg [was added](https://github.com/ros-perception/vision_msgs/commit/ef04cafde553780569bb53f0d0bd3d0edc04bfc4) to the development repo.

Could the maintainer of the release repo update the noetic release to include this message?

Kukanani, 2022-04-09:
I don't know why the documentation updated when I did not create a new release. Either way, here's the PR for the release to add the bounding box array messages. https://github.com/ros/rosdistro/pull/32721

igor-markovic, 2022-04-26:
`BoundingBox2DArray` is now a message when installing the latest version, thanks!

closed 2022-04-26 by igor-markovic

yeyanan93, 2023-12-12:
how can I solve the problem vai apt inatall?

sudo apt isntall ros-noetic-vision-msgs