ros-perception/vision_msgs #70: Ros2 Rolling has a strange Pose2D msgs

geoeo, 2022-05-06:
I just upgraded to ros2 rolling on a machine and my Pose2D msgs is different from galactic

New:
```
`class Pose2D(metaclass=Metaclass_Pose2D):
    """Message class 'Pose2D'."""

    __slots__ = [
        '_position',
        '_theta',
    ]

    _fields_and_field_types = {
        'position': 'vision_msgs/Point2D',
        'theta': 'double',
    }
```

As you can see this class has a position element which wraps x,y. These two values are supposed to be inline? Where does this come from. I cant see the change in the Humble branch?

SteveMacenski, 2022-05-06:
We did indeed add a new Pose2D message to this project starting in v4.0.0 which should be in Humble and Rolling (I checked too, it is released to both, but I can't speak if its been synced yet).

We did this based on feedback of abuse of that message time and because the plan is to deprecate the geometry_msgs version of it. While it is API the same, yes, it is in a new location so from a code auto-gen perspective, there is a change

geoeo, 2022-05-06:
I see, so this is intended and known that this breaks interfaces to galactic?

SteveMacenski, 2022-05-06:
Yeah

geoeo, 2022-05-07:
Alright. I just have to have strange catches in our code while we migrate to rolling/humble. Thanks!

closed 2022-05-07 by geoeo