ros-perception/vision_msgs #72: ROS 2 RVIZ plugin

NovoG93, 2022-06-29:
Hey guys, 
just wanted to share my RVIZ2 plugin to visiualize Detection3dArray msgs.
Will update it as described in the readme when I find time.

See here: [https://github.com/NovoG93/vision_msgs_rviz_plugins](https://github.com/NovoG93/vision_msgs_rviz_plugins)

Open for any suggestions if you have some.

Best

NovoG93, 2022-12-29:
Hey guys,
I ported my plugin to humble.

See here: [https://github.com/NovoG93/vision_msgs_rviz_plugins](https://github.com/NovoG93/vision_msgs_rviz_plugins)

Open for any suggestions if you have some.

Best

NovoG93, 2023-01-01:
Hey guys,
I added support to show the detection score and for [vision_msgs/Detection3D](https://github.com/ros-perception/vision_msgs/blob/noetic-devel/msg/Detection3D.msg). For a full list of supported features see: https://github.com/NovoG93/vision_msgs_rviz_plugins
[image]
Open for any suggestions if you have some.

Best

SteveMacenski, 2023-01-02:
Wow looks great!

NovoG93, 2023-01-22:
Hey everyone,

I added support for BoundingBox3D and BoundingBox3DArray msgs, for a full list of supported features see: https://github.com/NovoG93/vision_msgs_rviz_plugins:
[image]
Open for any suggestions if you have some.

Best

mintar, 2023-01-23:
Looks nice!

Since you asked, the only feedback that I have is that you seem to have hardcoded some class id to semantic names values ("car: orange, person: blue, cyclist: yellow, motorcycle: purple, other: grey"). These are specific to your application and probably not very useful in general. However, as there is no generic way to look up the class id to class name mapping, I don't have a better idea. :)

NovoG93, 2023-01-23:
@mintar I will look into ways of changing the color maps online. Maybe by providing the path to a yaml file.

SteveMacenski, 2023-01-23:
Some defaults probably makes sense, but also being able to change that would be nice if that profile is provided.

Let me know if you're open to contributing this here! I took a quick glance through the code and it looks pretty clean to me!

SteveMacenski, 2023-01-24:
A PR so if I run bloom, this can be released too!

closed 2023-01-31 by SteveMacenski

SteveMacenski, 2023-02-07:
@NovoG93 I left a note on https://github.com/ros-perception/vision_msgs/pull/79#issuecomment-1421429602 -- there's an issue with the rviz plugins compiling in isolation.

NovoG93, 2023-02-16:
@SteveMacenski will the rviz plugins also be available via the package manager (e.g. apt)?

SteveMacenski, 2023-02-16:
I tried to run a release last week but ran into an issue with bloom since I don't have push access to the release repository for this package. @Kukanani can you run a release? I already tagged it, bloom just needs to be run

NovoG93, 2023-02-17:
I see, thanks for the update

Kukanani, 2023-02-17:
Yes, I'll bloom. Hopefully can get to it this weekend.

We should reach out to Tully or someone else who can grant release repo access for others. Or is it because it's the older Foxy release, and it still points to the old release repo?

SteveMacenski, 2023-02-21:
I was given access over the weekend

NovoG93, 2023-02-24:
Although I can see that the vision_msgs_rviz_plugins are built on the master in the jenkins build [here](https://build.ros2.org/job/humble_rosdistro-cache/87693/console) and that it has been relesed [here](https://github.com/ros2-gbp/vision_msgs-release) and [here](http://repo.ros2.org/rosdistro_cache/) I can't seem to install it via apt.

Inside the docker container (osrf/ros:humble-desktop) I ran `rosdep update && apt update` and searched with `apt-cache search` but couldn't find the plugins. Does it simply take a little longer to be fully deployed?

output of `apt-cach search ros-humble-vision-msgs`:
```shell
root@c1f18678f1bd:/# apt-cache search ros-humble-vision-msgs
ros-humble-vision-msgs - Messages for interfacing with various computer vision pipelines, such as object detectors.
ros-humble-vision-msgs-dbgsym - debug symbols for ros-humble-vision-msgs
ros-humble-vision-msgs-layers - Collection of RQt Image Overlay Plugins for Vision Msgs
ros-humble-vision-msgs-layers-dbgsym - debug symbols for ros-humble-vision-msgs-layers
```

output of `apt-cache search rviz | grep plugin`:
```shell
root@c1f18678f1bd:/# apt-cache search rviz | grep plugin
ros-humble-grid-map-rviz-plugin - RViz plugin for displaying grid map messages.
ros-humble-grid-map-rviz-plugin-dbgsym - debug symbols for ros-humble-grid-map-rviz-plugin
ros-humble-nav2-rviz-plugins - Navigation 2 plugins for rviz
ros-humble-nav2-rviz-plugins-dbgsym - debug symbols for ros-humble-nav2-rviz-plugins
ros-humble-octomap-rviz-plugins - A set of plugins for displaying occupancy information decoded from binary octomap messages.
ros-humble-octomap-rviz-plugins-dbgsym - debug symbols for ros-humble-octomap-rviz-plugins
ros-humble-rmf-visualization-rviz2-plugins - A package containing RViz2 plugins for RMF
ros-humble-rmf-visualization-rviz2-plugins-dbgsym - debug symbols for ros-humble-rmf-visualization-rviz2-plugins
ros-humble-rviz-2d-overlay-plugins - RViz2 plugin for 2D overlays in the 3D view.
ros-humble-rviz-2d-overlay-plugins-dbgsym - debug symbols for ros-humble-rviz-2d-overlay-plugins
ros-humble-rviz-common - Common rviz API, used by rviz plugins and applications.
ros-humble-rviz-default-plugins - Several default plugins for rviz to cover the basic functionality.
ros-humble-rviz-default-plugins-dbgsym - debug symbols for ros-humble-rviz-default-plugins
ros-humble-rviz-imu-plugin - RVIZ plugin for IMU visualization
ros-humble-rviz-imu-plugin-dbgsym - debug symbols for ros-humble-rviz-imu-plugin
ros-humble-rviz-rendering-tests - Example plugin for RViz - documents and tests RViz plugin development
ros-rolling-octomap-rviz-plugins - A set of plugins for displaying occupancy information decoded from binary octomap messages.
ros-rolling-octomap-rviz-plugins-dbgsym - debug symbols for ros-rolling-octomap-rviz-plugins
ros-rolling-rmf-visualization-rviz2-plugins - A package containing RViz2 plugins for RMF
ros-rolling-rmf-visualization-rviz2-plugins-dbgsym - debug symbols for ros-rolling-rmf-visualization-rviz2-plugins
ros-rolling-rviz-2d-overlay-plugins - RViz2 plugin for 2D overlays in the 3D view.
ros-rolling-rviz-2d-overlay-plugins-dbgsym - debug symbols for ros-rolling-rviz-2d-overlay-plugins
ros-rolling-rviz-common - Common rviz API, used by rviz plugins and applications.
ros-rolling-rviz-default-plugins - Several default plugins for rviz to cover the basic functionality.
ros-rolling-rviz-default-plugins-dbgsym - debug symbols for ros-rolling-rviz-default-plugins
ros-rolling-rviz-imu-plugin - RVIZ plugin for IMU visualization
ros-rolling-rviz-imu-plugin-dbgsym - debug symbols for ros-rolling-rviz-imu-plugin
ros-rolling-rviz-rendering-tests - Example plugin for RViz - documents and tests RViz plugin development
```

Kukanani, 2023-02-24:
There is another step to go from having the ROS packages building in Jenkins to having them available in the public repositories. This is called a Release Sync and it happens every few months. Syncs are announced on the ROS Discourse by the OSRF team.

NovoG93, 2023-02-24:
Thank you for explaining that for me!