ros-perception/vision_msgs #76: ObjectHypothesisWithPose with label

gbr1, 2022-09-03:
Hi,
I'm working on wrapping [edgeimpulse on ROS2](https://github.com/gbr1/edgeimpulse_ros).

It could be useful to have a `string label` parameter in the message type `ObjectHypothesisWithPose` because it often happen that you want to send also the label.

What do you think about that?

Thank you in advance!

gbr1, 2022-09-04:
I found another issue:
- **id** in `ObjectHypothesisWithPose` is numeric (int64) for noetic but for foxy is a string
- in **galactic** ObjectHypothesis is used instead

Please can you make unified?
Thank you in advance

SteveMacenski, 2022-09-04:
That change was intentional

gbr1, 2022-09-04:
@SteveMacenski so should be foxy updated to the galactic message version (the one with ObjectHypothesis)?

zkytony, 2022-09-12:
I noticed the `id` field was once changed to string type, but then reverted (see https://github.com/ros-perception/vision_msgs/commit/fd30ff65e2a5a65353dcf8c35838666d5efbe444). What happened? It would be really convenient to have the `id` field be a string.

Kukanani, 2022-09-15:
We don't usually unify message definitions across distros unless there is a very strong reason to do so.  Because ROS messages are very brittle (non-identical message definitions cause all kinds of headaches), making breaking changes usually only happens between distros.

Future distros will all have `ObjectHypothesis` as the `id` field (unless another new change comes in).