ros-perception/vision_msgs #78: Using vision_msgs with ROS2 Humble

Tamire, 2022-12-25:
I have been having some difficulties integrating vision_msgs to my ROS2 project(it worked great on ROS1).

After cloning and checking out the ROS2 project, I was able to build vision msgs(located on ros2_ws/src).

To my Package.xm, I added: `  <depend>vision_msgs</depend>`

And  to CmakeLists.txt: 
`find_package(vision_msgs REQUIRED)
ament_target_dependencies(${PROJECT_NAME}
        ...
        vision_msgs
        )
`
and even to my target_link_libraries.

When trying to include `#include <vision_msgs/Detection2DArray.h>` I am getting an error saying it cannot be located.

I also tried to remove my dependent code and simply build it with the dependency, but getting(even after source install/setup.bash):

> /usr/bin/ld: cannot find -lvision_msgs: No such file or directory

Any help would be greatly appreciated.

Thanks!

SteveMacenski, 2022-12-26:
The header format is different. Its now totally snake case for the message header file definitions under the msg directory

Tamire, 2023-01-08:
As @SteveMacenski mentioned, all headers are now snake cased.
If this helps anyone in the future, use:
`#include <vision_msgs/msg/detection2_d_array.hpp>`

closed 2023-01-08 by Tamire