ros-perception/vision_msgs #80: class std::basic_ostream<char>’ has no member named ‘str’

gorghino, 2023-02-01:
I tried to compile vision_msgs in my ROS2 Humble workspace but I got this error while it was building vision_msgs_rviz_plugins:

```
--- stderr: vision_msgs_rviz_plugins                                                                                                                                                                 
In file included from /home/nvidia/drone-racing/build/vision_msgs_rviz_plugins/include/vision_msgs_rviz_plugins/../../../../src/vision_msgs/vision_msgs_rviz_plugins/include/vision_msgs_rviz_plugins/detection_3d_array.hpp:33,
                 from /home/nvidia/drone-racing/build/vision_msgs_rviz_plugins/include/vision_msgs_rviz_plugins/moc_detection_3d_array.cpp:9:
/home/nvidia/drone-racing/src/vision_msgs/vision_msgs_rviz_plugins/include/vision_msgs_rviz_plugins/detection_3d_common.hpp: In member function ‘void rviz_plugins::Detection3DCommon<MessageType>::ShowScore(vision_msgs::msg::Detection3D, double, size_t)’:
/home/nvidia/drone-racing/src/vision_msgs/vision_msgs_rviz_plugins/include/vision_msgs_rviz_plugins/detection_3d_common.hpp:450:90: error: ‘std::basic_ostream<char>::__ostream_type’ {aka ‘class std::basic_ostream<char>’} has no member named ‘str’
  450 |     marker->text = (std::ostringstream{} << std::fixed << std::setprecision(2) << score).str();
      |                                                                                          ^~~
make[2]: *** [CMakeFiles/vision_msgs_rviz_plugins.dir/build.make:128: CMakeFiles/vision_msgs_rviz_plugins.dir/include/vision_msgs_rviz_plugins/moc_detection_3d_array.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:138: CMakeFiles/vision_msgs_rviz_plugins.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

```

https://github.com/ros-perception/vision_msgs/blob/ros2/vision_msgs_rviz_plugins/include/vision_msgs_rviz_plugins/detection_3d_common.hpp#L450

I fixed it dividing the one-line command into multiple <<

```
    std::ostringstream oss;
    oss << std::fixed;
    oss << std::setprecision(2);
    oss << score;
    marker->text = oss.str();
```

SteveMacenski, 2023-02-01:
If you file a PR, I’ll happily merge.  Might be a platform quirk on ARM

SteveMacenski, 2023-02-03:
I had to revert your PR due to build failures https://github.com/ros-perception/vision_msgs/pull/82

Tamire, 2023-02-07:
Hi There,
I'm having the same issue(Ubuntu 22). Any update on this topic?

gorghino, 2023-02-07:
@Tamire https://github.com/ros-perception/vision_msgs/pull/83 shall fix it. Could you test?

Tamire, 2023-02-07:
@gorghino Works for me. hopefully wont fail build now!

closed 2023-02-07 by SteveMacenski