ros-perception/vision_msgs #88: Cannot find vision_msgs/msg/detail/bounding_box2_d__struct.hpp

jainishmehta, 2023-05-15:
HI,
I am having a similar issue with this:
/home/jainish/ros2_ws/build/webots_ros2_msgs/rosidl_generator_cpp/webots_ros2_msgs/msg/detail/camera_recognition_object__struct.hpp:23:10: fatal error: vision_msgs/msg/detail/bounding_box2_d__struct.hpp: No such file or directory
23 | #include "vision_msgs/msg/detail/bounding_box2_d__struct.hpp"

The other packages like geometric_msgs works fine though.

Kukanani, 2023-05-15:
Hi, sorry to hear you're having trouble. Could you please provide reproduction steps? What version of ROS are you running, what command are you running that errors out, what are you trying to build (looks like `webots_ros2_msgs`, what version)?

jainishmehta, 2023-05-15:
https://github.com/cyberbotics/webots_ros2/wiki/Linux-Installation-Guide

Yeah, it is from webots_ros2_msg. If you follow the instructions on the link. WHere you first 

1. mkdir -p ~/ros2_ws/src
2. cd ~/ros2_ws
git clone --recurse-submodules https://github.com/cyberbotics/webots_ros2.git src/webots_ros2
3. sudo apt install python3-pip python3-rosdep python3-colcon-common-extensions
sudo rosdep init && rosdep update
rosdep install --from-paths src --ignore-src --rosdistro $ROS_DISTRO
4. colcon build

The colcon build is where it gets stuck n it. Do I need to redownload the package for vision_msgs, I'm not sure. 

Thanks

SteveMacenski, 2023-05-15:
Is that IDL generated code for the messages or your actual code? if its not your actual code, link the msgs file you're using. If this is your file, why are you using internal detail struct includes instead of the normal messages? This also looks like you're using Crystal, which is ancient and could very well not include necessary types or features that this package leverages.

SteveMacenski, 2025-05-29:
Closing due to inactivity

closed 2025-05-29 by SteveMacenski