ros-perception/vision_msgs #89: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >

kenloumixx, 2023-05-22:
Though this question is not directly related to the vision_msgs, I hope someone here might know the answer for this error. 

Hello,
I build ROS1 noetic and ROS2 galactic both from source.
I am facing the below error while building ROS1_bridge:

```
Starting >>> ros1_bridge
[Processing: ros1_bridge]                              
[Processing: ros1_bridge]                                      
--- stderr: ros1_bridge                                          
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >, vision_msgs::msg::Detection3D_<std::allocator<void> > >::convert_1_to_2(vision_msgs::Detection3D_<std::allocator<void> > const&, vision_msgs::msg::Detection3D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection2D_<std::allocator<void> >, vision_msgs::msg::Detection2D_<std::allocator<void> > >::convert_2_to_1(vision_msgs::msg::Detection2D_<std::allocator<void> > const&, vision_msgs::Detection2D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection2D_<std::allocator<void> >, vision_msgs::msg::Detection2D_<std::allocator<void> > >::convert_1_to_2(vision_msgs::Detection2D_<std::allocator<void> > const&, vision_msgs::msg::Detection2D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >, vision_msgs::msg::Detection3D_<std::allocator<void> > >::convert_2_to_1(vision_msgs::msg::Detection3D_<std::allocator<void> > const&, vision_msgs::Detection3D_<std::allocator<void> >&)'
collect2: error: ld returned 1 exit status
make[2]: *** [dynamic_bridge] Error 1
make[1]: *** [CMakeFiles/dynamic_bridge.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >, vision_msgs::msg::Detection3D_<std::allocator<void> > >::convert_1_to_2(vision_msgs::Detection3D_<std::allocator<void> > const&, vision_msgs::msg::Detection3D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection2D_<std::allocator<void> >, vision_msgs::msg::Detection2D_<std::allocator<void> > >::convert_2_to_1(vision_msgs::msg::Detection2D_<std::allocator<void> > const&, vision_msgs::Detection2D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection2D_<std::allocator<void> >, vision_msgs::msg::Detection2D_<std::allocator<void> > >::convert_1_to_2(vision_msgs::Detection2D_<std::allocator<void> > const&, vision_msgs::msg::Detection2D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >, vision_msgs::msg::Detection3D_<std::allocator<void> > >::convert_2_to_1(vision_msgs::msg::Detection3D_<std::allocator<void> > const&, vision_msgs::Detection3D_<std::allocator<void> >&)'
collect2: error: ld returned 1 exit status
make[2]: *** [parameter_bridge] Error 1
make[1]: *** [CMakeFiles/parameter_bridge.dir/all] Error 2
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >, vision_msgs::msg::Detection3D_<std::allocator<void> > >::convert_1_to_2(vision_msgs::Detection3D_<std::allocator<void> > const&, vision_msgs::msg::Detection3D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection2D_<std::allocator<void> >, vision_msgs::msg::Detection2D_<std::allocator<void> > >::convert_2_to_1(vision_msgs::msg::Detection2D_<std::allocator<void> > const&, vision_msgs::Detection2D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection2D_<std::allocator<void> >, vision_msgs::msg::Detection2D_<std::allocator<void> > >::convert_1_to_2(vision_msgs::Detection2D_<std::allocator<void> > const&, vision_msgs::msg::Detection2D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >, vision_msgs::msg::Detection3D_<std::allocator<void> > >::convert_2_to_1(vision_msgs::msg::Detection3D_<std::allocator<void> > const&, vision_msgs::Detection3D_<std::allocator<void> >&)'
collect2: error: ld returned 1 exit status
make[2]: *** [static_bridge] Error 1
make[1]: *** [CMakeFiles/static_bridge.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< ros1_bridge [1min 19s, exited with code 2]

Summary: 0 packages finished [1min 20s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

I build ROS1 and ROS2 both from souce, and build each vision_msgs in both versions. 
TIA

SteveMacenski, 2023-05-22:
They do not directly correlate to do a direct bridge conversation I think is what you're learning here. Vision msgs are relatively speaking to other msgs packages new and are thus more subject to non-trivial change

Kukanani, 2023-05-25:
Agreed, this seems similar to #39 (although errors are happening at link time instead of compile time)

closed 2025-05-29 by SteveMacenski