ros-perception/vision_msgs #90: Replace Pose2D with geometry_msgs/msg/Pose2D

sloretz, 2023-06-05:
It looks like `vision_msgs/msg/Pose2D` and `geometry_msgs/msg/Pose2D` are identical, or at least they were before https://github.com/ros-perception/vision_msgs/pull/67 split it into Point2D and Pose2D (where Point2D is used only by Pose2D).

* https://github.com/ros-perception/vision_msgs/blob/ros2/vision_msgs/msg/Pose2D.msg
* https://github.com/ros2/common_interfaces/blob/rolling/geometry_msgs/msg/Pose2D.msg

I propose removing Point2D and Pose2D from this repo and using `geometry_msgs/msg/Pose2D` instead.

SteveMacenski, 2023-06-05:
The depreciation notice in Geometry Messages is why its not used. I've been advocating against its eventual removal for some time now due to things like this (among others). If that planned removal is withheld, then I'd be more than happy to use the geometry_msgs  version.  

https://github.com/ros-perception/vision_msgs/pull/64

mintar, 2023-06-07:
These two messages are not semantically identical, so they have to stay separate:

* geometry_msgs/Pose2D is in meters
* vision_msgs/Pose2D is in pixels