ros-perception/vision_msgs #105: Source_Cloud field missing from detection3D

marcomasa, 2025-05-07:
Hi!

I noticed that the source_cloud field is missing now in the detection3D message template.
Is there an alternative now?

I cannot find relevant information in #43 or in the remaining messages, so I wonder if there has been some replacement added.

Thanks for your help!

SteveMacenski, 2025-05-07:
What is the source_cloud for in this case? Is this the entire PC2 that the bounding box comprises? I don't think that's the role of this message definition, but open for feedback about future changes. I'm not sure what you mean by now missing, I don't see that we've ever had that

marcomasa, 2025-05-07:
I have the use case where I fill the Detection3D array using a 3D camera and semantic segmentation.
I have the setup where 

Perception  --> vision_msgs/Detection3D --> World Model 

are connected, but multiple perception inputs could contribute to the same objects in the world model. 

I would like to send the cloud that belongs to a specific detection together with the result, in order to e.g. construct a full view of an object from two perspectives. I would only send the points of the cloud that show the object, not the full source cloud.
The size of the cloud could obviously vary in that case, but its usually much smaller than a full cloud.

It used to be available in the [ROS1 Message Definition](https://docs.ros.org/en/kinetic/api/vision_msgs/html/msg/Detection3D.html):

```
File: vision_msgs/Detection3D.msg
------------------------------------------------------------------------------------

# Defines a 3D detection result.
#
# This extends a basic 3D classification by including position information,
#   allowing a classification result for a specific position in an image to
#   to be located in the larger image.

Header header

# Class probabilities. Does not have to include hypotheses for all possible
#   object ids, the scores for any ids not listed are assumed to be 0.
ObjectHypothesisWithPose[] results

# 3D bounding box surrounding the object.
BoundingBox3D bbox

# The 3D data that generated these results (i.e. region proposal cropped out of
#   the image). This information is not required for all detectors, so it may
#   be empty.
sensor_msgs/PointCloud2 source_cloud

``` 

Ideally I would like to make use of the RVIZ vision_msgs plugin to publish and display the message directly.
So a custom message would not be ideal for me.

Is it planned to add this feature back in?

SteveMacenski, 2025-05-07:
I'm on the go today otherwise I'd do this for you and make a summary -- but can you go through the git history and find where it was removed? I'm wondering if it was accidental or intentional. If intentional, what the motivation was. I'm seeing [the comment](https://github.com/ros-perception/vision_msgs/blob/ros2/vision_msgs/msg/Detection3D.msg#L20-L23) for the source cloud but not the field itself, which is eyebrow raising.

https://github.com/ros-perception/vision_msgs/commits/

marcomasa, 2025-05-07:
Seems like it was [this commit](https://github.com/ros-perception/vision_msgs/commit/163d5389d6df1271688e2ce12f7d6c0ff3296ad8#diff-34732dae0e5e2d3eaeebec9e9e0386d49fea09b288e0d4758ecc7b5fa66401a9R19)

Discussion happened in #53 and led to this:

[image]

Makes me wonder if support in RVIZ also got dropped.
The [Plugin](https://github.com/ros-perception/vision_msgs/blame/ros2/vision_msgs_rviz_plugins/src/detection_3d.cpp) also doesn't appear to support visualizing the pcl.

The [ROS1 Branch](https://github.com/ros-perception/vision_msgs/tree/noetic-devel) seems to not have RVIZ Plugins at all.
So I guess it would be a new feature?

mintar, 2025-05-08:
Seems like you found the relevant discussions! We discussed this indeed in length in #43, and decided to remove it, which was done in #53. That PR also added the rationale to the README:

> Source data that generated a classification or detection are not a part of the messages. If you need to access them, use an [exact or approximate time synchronizer](http://wiki.ros.org/message_filters#Policy-Based_Synchronizer_.5BROS_1.1.2B-.5D) in your code, as the message's header should match the header of the source data.

The longer justification was also quoted there:

> Decouple from data: Some messages here include a field carrying some source data. I understand it is very practical on one hand (we do it too in our custom messages), but on the other hand it generates issues like https://github.com/ros-perception/vision_msgs/issues/20 (and possibly future issues like sensor_msgs/PointCloud support etc.). And these have no good solution with the data inside the message except for composition on a higher level. I propose to remove the source data fields from vision_msgs, and possibly leave a note in the message definition that the user should either use a message synchronizer to get the pairs of image-detection, or create a custom type that contains both the image (cloud...) and the detection message. It's easy to use topic_tools/relay_field to extract the desired parts afterwards. There are situations where you do not want to relate the detection to the whole image, but you want to e.g. have the cutout from the image where the detection occured. This would work the same in case 1 detection per image is published (you'd just publish the cutout on a new topic). But it would become a bit more complicated in case the detector can return multiple detections per image and you would use DetectionArray. Here, matching by header would no longer be enough. Or - if taken rigorously - it could be, because making a cutout from an image formally invalidates its frame_id. So each cutout should have its own frame_id, and then you could make the detection-image mapping. Some more rationale: someone might want to run detections on an organized 3D pointcloud from a depth camera and return 2D detections (because he's e.g. only interested in bounding boxes in the RGB in the end). And a practical issue we faced: on a low-bandwidth link, we wanted to send detections, but as I mentioned earlier, the image is already a part of our messages (similarly to how it is done here now). But we figured out we do not want to transmit the image over the slow link. So we ended up not filling any values in the image fields, and now we have detections which lie to be detected on 0x0 px images. That doesn't really seem like a conceptual approach. Still, I'm not really 100% sure on this point, but it seems to me decoupling is the "more correct" way, but has its pros and cons. Bad thing is that using composition to create a DetectionsWithImagesArray message would not subsequently allow to use topic_tools/relay_field, because you would need to relay primitive-array fields, which I guess is not supported.

marcomasa, 2025-05-08:
I can follow the discussion to some extend. However, it mentioned exactly what my issue with the current solution is:
Matching multiple detections (Detection3DArray - D3A) to parts of the PCL. 

My option now would be to either publish all the PCL segments over the same topic, and then having to "manually" match them to the D3A, or creating a custom message type which combines something like D3A and PCL Array..

However, both solutions would not allow me to visualize both the PCL points and the Bounding Box of all detections at the same time.

I am currently working on a forked version of this repo, where I just added the field back in.
It just seems easier for my setup on the receiving end (world model).
Visualization in RVIZ is not supported atm tho, I create and publish an additional PCL combining all the segments.

SteveMacenski, 2025-05-08:
The time synchronization case would allow you to do that - moreso you'd be able to publish the PC2's as PC2s on the ROS network to be able to immediately visualize in rviz for debugging without needing to write a new display plugin type for the visualization cloud. 

If you would like this for another application (world models), maybe the right answer is to discuss a new message type for world models to communicate its information without impacting the other 3D vision based messages?

mintar, 2025-05-08:
There are many reasons why you wouldn't want the source data in the Detection2D / Detection3D messages:

1. blows up the message size tremendously
2. doesn't support compression
3. assumes that Detection3D comes from point clouds and Detection2D comes from images; however, modern 6DoF pose estimation approaches usually derive Detection3D from RGB images or RGB-D image pairs, and those weren't supported anyway.

In short, we cannot just add source point clouds, images, RGB-D images etc. back to the messages here.

> My option now would be to either publish all the PCL segments over the same topic, and then having to "manually" match them to the D3A

I wouldn't do this, because the Time Synchronizer can only do 1:1 sync between two topics. It cannot match 1 message on topic A to N messages on topic B.

> or creating a custom message type which combines something like D3A and PCL Array..

Yes, that would be an option.

Option 3: Use a Time Synchronizer between the Detection3DArray topic and the original (full) point cloud topic, then cut the objects out of the point cloud using the 3D bounding boxes.

Option 4: Create a custom message type which is basically a list of object point clouds, then use a Time Synchronizer to sync it with the Detection3DArray topic. But there are probably no benefits of this option over option 2.

Kukanani, 2025-05-09:
Weighing in here - without knowing your exact node setup, I think @mintar 's Option 3 would be a flexible solution that represents a good abstraction. You have your raw sensor data, and you have some interpretation of that data. There may be multiple interpretations, but they all are derived from the same data.

One change I might make (option 3b?) is to actually pass along the timestamp of the original 3D data along with your detection - either by matching the Detection3D's timestamp to the original data (as suggested [here](https://github.com/ros-perception/vision_msgs/blob/21b2d3ff0f8e95b1330dca87a21aea1cc3feda58/vision_msgs/msg/Detection3D.msg#L22)), or, if you really need both the time of the source data, and the time the detections were produced, by wrapping in another message that contains a Detection3D and another header.

mintar, 2025-05-09:
I believe that message headers should *always* copy the time stamp of the original data. Otherwise, libraries like TF don't work. It's fine to have another time stamp (when the message was processed) if it's really relevant, but not in the header.

marcomasa, 2025-05-09:
Thank you all for your contributions! :)

Option 3 is not really an option for me unfortunately, since I am not using the original PCL from the camera, but rather construct the object segments based on rgb + depth images and the segmented pixels.
So at that point I already have the cutouts available, which I use to even create the Bounding Box. 
Doing the cutout again in the sync subscriber seems very inefficient and potentially error-prone to me.'

Also this whole merging thing with timestamps would really not work I think. 
I'd have, as @mintar mentioned, one message with type Detection3DArray and multiple ones with PCL2,
all having the same stamp. 
In addition, matching the cutouts to the detections afterwards also does not seem efficient to me.

So to summarize, I think using some custom message containing all data makes the most sense in my application.

My tests with the custom vision_msgs unfortunately have not been too successful yet.
RVIZ and/or my custom node crash when the message contains the additional "source_cloud". 
I dont exactly know why, but I assume its kinda related to what @SteveMacenski mentioned with me needing to write 
my own plugin for the adapted version.

So atm, I need to re-consider how I design the architecture a bit.
I will let you know once I have found a proper solution. 

Maybe a question:
Would it make sense for me, if I now implement an additional message type that includes the source data (lets say "Detection3DwithSources") to make a PR?
It would add the dependencies back in that were removed (I'd say rightfully at the time),
but extend the functionalities for other people that have a similar use case like me.
It would also keep the more lightweight Detection3D message as it is now. 
What do you think about it? :)

PS: 
I keep the timestamp of the original sensor data. I agree that it should be kept that way for TF to work properly.
Additionally, i need to know when the detection happened in the real world in order to merge different modalities in further processing steps (world model).

SteveMacenski, 2025-05-09:
>  Would it make sense for me, if I now implement an additional message type that includes the source data (lets say "Detection3DwithSources") to make a PR?

Possibly! I think Being more semantic in application would be important. If you're using them for something in particular, chances are so is someone else, and we should name it appropriately by application rather than its data contained.

marcomasa, 2025-05-12:
Hi guys!

I've spent some time developing my custom solution now, which seems to work so far.

Basically I've added two message types
- Detection3DWithPcl.msg
- Detection3DWithPclArray.msg

and the RVIZ Plugins for it.
You can check out [my fork](https://github.com/marcomasa/vision_msgs) for it.

A little preview of how it looks in RVIZ using the adapted 'samples' files:

[image]